DiamondLightSource/dodal #716: Convert the OAV to ophyd-async

DominicOram, 2024-07-29:
We should try and get all our devices into `ophyd-async`. The OAV is one of the last ones left

## Note
This is a reasonably large device, it may make sense to split it into multiple issues

## Acceptance Criteria
* The OAV is converted into `ophyd-async`
* Tests all still pass

DominicOram, 2024-09-24:
https://github.com/DiamondLightSource/dodal/issues/801 should be done first

DominicOram, 2024-09-24:
As part of this we should probably look at how we do the `OAVConfigParams`. Particularly how they get updated from the device. It may make more sense for them to be derived signals https://github.com/bluesky/ophyd-async/issues/525

noemifrisina, 2024-09-26:
As this is a fairly large device and we probably can't have an ophyd-async device with ophyd components (or viceversa), the first step will be to make a new very basic ophyd-async device meeting the minimum needs for i24 serial and then we can add the rest of the functionality as we go along.

PR #857 (2024-11-04): Finish setting up an ophyd_async OAV
Closes #716 
Closes #824 

- Create an ophyd_async MJPG area detector plugin to replace old one
-  Re-write snapshot devices to inherit from new MJPG
- Add snapshots to ophyd_async OAV
- Use async OAV in beamline device instantiation
- Add a small CAM plugin with minimum requirements for hyperion
- Add unit tests, fix system tests to work with async device
- Remove all obsolete code and tests

### Instructions to reviewer on how to test:
1. Check that all the functionality of the old ophyd OAV device and MJPG plugin is covered by the new ophyd_async code
2. [... cut]