DiamondLightSource/dodal #724: Create an ophyd device for Applet2 undulator

Relm-Arrowny, 2024-08-05:
I17 will be using the same type of ID as I10, I21 and I06. This issue is to create an ophyd_async device to control this type of ID.
This type of ID has 4 extra degrees of freedom beside the gap, each bank of magnet can move independently to each other. To produce different polarisation as well as energy.

User expected to be able to set energy and polarisation states: PC, NC, LH, LV and LA. With LA there will also be a polarisation angle associated with this state.
These polarisation states are achieved by moving the ID motors into its correct position using a lookup table, each polarisation has their own look up table, therefore changing polarisation is effectively changing look up table.

## Acceptance Criteria
Create an applet2 id device that can move all the id's motors. similar to [undulator](src/dodal/devices/undulator.py)
Must connect all required PV.
Passes tests.

Relm-Arrowny, 2024-08-05:
@DominicOram @stan-dot I saw the id issue on i18 and wondering if you two have any tips and comment on this? Especially, I want to make sure no such device already exist and any advice on best/standard way to implement the extra lookup table/ x-ray polarisation would be much appreciated.

stan-dot, 2024-08-05:
@Relm-Arrowny  for me this link is broken: 
``` Create an applet2 id device that can move all the id's motors. similar to [undulator](https://github.com/DiamondLightSource/dodal/issues/src/dodal/devices/undulator.py)

which i18 id issue?

Relm-Arrowny, 2024-08-05:
[this](https://github.com/DiamondLightSource/dodal/issues/706) Saw you two discussing undulator and thought I ask :p.

DominicOram, 2024-08-07:
You may be able to re-use https://github.com/DiamondLightSource/dodal/pull/721 but it depends on the exact logic. Is it that:
* Changing the energy still only changes the gap demand PV
* The lookup table used for this demand is different and will change based on the polarisations
* Changing the polarisations changes this lookup table and changes some of the phase demand positions too

Or is it that:
* Changing the energy will change the gap demand and phase demand PVs
* How these change are based on lookup tables
* These tables are set based on polarisation

If it's the first I think you can reuse the existing undulator quite a bit, if it's the second I would write an ABC that they share so that we can use them interchangeably but otherwise have two separate devices.

PR #744 (2024-10-23): 724 create an ophyd device for apple 2 undulator and add i10 as a beamline
Fixes #724 

### Instructions to reviewer on how to test:
1. Confirm connection.
2. Confirm test pass.

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the rel [... cut]
closed 2024-10-23 by Relm-Arrowny