DiamondLightSource/dodal #734: Make a redis lookup table caching device

stan-dot, 2024-08-08:
A brief description of the issue, if it's for a specific beamline say which beamline and why they want it

As per discussion with @callumforrester  and @DiamondJoseph this is needed for db IO as this cannot be in a plan or a stub as it would clog the RunEngine

## Acceptance Criteria
* I can spin the device up at i18 and make a quick test plan to test redis read and write

callumforrester, 2024-08-08:
@stan-dot do you plan to put an API between the device and redis?

stan-dot, 2024-08-19:
what do you mean? I'd use the redis client library and point at the redis instance on the cluster

stan-dot, 2024-08-19:
todo: need to move the temporary implementation from the i18-bluesky repo

stan-dot, 2024-08-30:
as @DominicOram commented elsewhere it fits best to include this as a child device in Undulator.

also this is relevant
https://github.com/DiamondLightSource/i18-bluesky/pull/14

stan-dot, 2024-08-30:
as the scan also uses data from the diode, it remains to be decided how to split the logic between the plan and the device. we could have a data structure to get from the diode, and do all the calculations from the device

closed 2024-09-17 by stan-dot