DiamondLightSource/dodal #759: Pressure jump devices collected updates

stan-dot, 2024-08-21:
Make sure that what is to be recorded is recorded and what is not is not

point 1 - limits and an error state if they are exceeded
            # todo double check if needed
            # self.pump_forward_limit = epics_signal_r(
            #     LimitSwitchState, prefix + "D74IN1"
            # )
            # self.pump_backward_limit = epics_signal_r(
            #     LimitSwitchState, prefix + "D74IN0"
            # )
point 2 - movement in manual mode
            self.pump_move_forward = epics_signal_rw(int, prefix + "M1:FORW")
            self.pump_move_backward = epics_signal_rw(bool, prefix + "M1:BACKW")

point 3 - do we need to reset the motor control? the `M1:CON`, if so, with `epics_signal_x`

point 4 - support arm and disarm for the fast valves

point5 - consider the openseq channel
        self.valves_open: DeviceVector[SignalR[int]] = DeviceVector(
            {i: epics_signal_r(int, f"{prefix}V{i}:OPENSEQ") for i in [1, 3, 5, 6]}
        )
or can we do everything through `CON`?

point 6- 
deleting gotobusy and other control signals not used in the acquisitions - can we confirm this?
            self.control_gotobusy = epics_signal_r(BusyState, prefix + "CTRL:GOTOBUSY")
            self.control_timer = epics_signal_r(TimerState, prefix + "CTRL:TIMER")
            self.control_counter = epics_signal_r(float, prefix + "CTRL:COUNTER")
            self.control_script_status = epics_signal_r(str, prefix + "CTRL:RESULT")
            self.control_routine = epics_signal_r(str, prefix + "CTRL:METHOD")
            self.control_iteration = epics_signal_r(float, prefix + "CTRL:ITER")

stan-dot, 2024-09-25:
@tpoliaw I know those are rough notes, does any of the 6 question have an obvious answer or are all exploratory?

tpoliaw, 2024-09-25:
1. I don't think GDA looked at limits 
2. GDA did nothing with manual control
3. I don't think we did in GDA
4. yes
5. GDA used `OPENSEQ` for opening the valves and `CON` for closing and resetting them. I can't remember why it made this distinction now.  Either way, I think an array of valves, each with an open and close signal would be better than a pair of arrays, one for all the open signals one for close signals.
6. `GOTOBUSY` was used in the Java implementation so that methods could block until the action had actually started (eg setPressure has a slight delay between hitting go and the pumps moving). If that is not needed in ophyd it can probably go. None of the other PVs were used.

stan-dot, 2024-09-25:
fantastic, thanks!

stan-dot, 2024-11-14:
this got solved with #812

PR #812 (2024-10-18): 759 pressure jump devices collected updates
Fixes #759 

The OPENSEQ PVs expect their value to be momentarily be set to 1 to trigger the valve opening `ValveOpenSeqRequest.OPEN_SEQ`. Via a synoptic screen this is achieved via a pushbutton control that is 1 when it is being held down and 0 when released, what is the equivalent way of doing this in dodal? can it be in the driver or is it down to the plan?

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsou [... cut]
closed 2024-11-14 by stan-dot