DiamondLightSource/dodal #761: Define standard baseline list to prevent re-definition per-plan

DiamondJoseph, 2024-08-22:
Our plans will begin to accept a `list[Device]` (`Readable`?) named `baseline` that are to be read at the start/end of a `run` in a stream named `baseline`.

These devices are ancillary devices that are not explicitly included in the scan logic (e.g. upstream optical components). They are likely to be consistent across multiple plans, although they may be overridden by explicitly passed arguments. 

Define a `list[Device|Readable]` either in the dodal module or in this repository that can be re-used as the default argument for this parameter.

DiamondJoseph, 2024-08-22:
https://github.com/DiamondLightSource/i22-bluesky/blob/1f0a00828b006388d1a56b0f426610ab8d7bfcdf/src/i22_bluesky/plans/stopflow.py#L59

Here is an example of how this was working in the i22-bluesky plans repository

stan-dot, 2024-10-23:
>  that are not explicitly included in the scan logic (e.g. upstream optical components). They are likely to be consistent across multiple plans, although they may be overridden by explicitly passed arguments.

@DiamondJoseph I think now with #841 merged we can look into this again. I am not sure I can see the full picture at this point.

in the i22 files

```python
devices = {flyer, panda} | detectors | baseline

    # Collect metadata
    plan_args = {
        "pre_stop_frames": pre_stop_frames,
        "post_stop_frames": post_stop_frames,
        "exposure": exposure,
        "shutter_time": shutter_time,
        "panda": panda.name + ":" + repr(panda),
        "detectors": {device.name + ":" + repr(device) for device in detectors},
        "baseline": {device.name + ":" + repr(device) for device in baseline},
    }
    
    ```
    
we have explicit inclusion of the baseline devices into the metadata logic.

Do you see the 'add baseline' as a new decorator? Or a change to all the plans? or a plan-decorator that adds 'scan baseline at the start' , `do plan`, 'scan baseline at the end'. And what about data writing? do we just write this out normally to the hdf files?

DiamondJoseph, 2024-10-23:
https://github.com/DiamondLightSource/i22-bluesky/blob/1f0a00828b006388d1a56b0f426610ab8d7bfcdf/src/i22_bluesky/plans/stopflow.py#L232

Bluesky already has a baseline pre-processor decorator that does all of the required changes to the plan procedure; you can see how in the linked plan it's used as a decorator to the inner plan. It is still an arg of the plan so I think it belongs in plan_args, but I'm not certain.

stan-dot, 2024-11-26:
@DominicOram do you need this at MX?

DominicOram, 2024-11-26:
In the long term probably yes but it's not a priority for us at the moment. I think we would probably just pick up any implementation you do for elsewhere, I can't imagine our usecases being very different

stan-dot, 2024-11-27:
@iain-hall would this feature be useful for our spectroscopy work?

iain-hall, 2024-11-27:
Yes - it sounds like standard set of metadata. Is it somehow different from metadata that would normally be included in a plan or set on runengine?

DiamondJoseph, 2024-11-27:
@iain-hall the metadata that gets passed into the plan is a dict with some pre-determined values:

`RE(plan(args, md={"sample": "The good biscuits"}))`

which adds into the StartDocument that is created `"sample": "The good biscuits"`

The baseline is a collection of devices to read at the start/end of the plan:

`RE(plan(args, baseline=[slits_1, slits_2, fswitch]))`

Which inserts `yield from trigger_and_read(*baseline, stream_name="baseline")` after `yield from open_run()` and before `yield from close_run()`

and leads to the creation of EventDocuments with data from all those devices.

iain-hall, 2024-11-27:
Thanks @DiamondJoseph  that makes sense. So 'metadata' passed via 'md' is just key-value pairs of strings, whereas the 'baseline' is for readout of device positions etc. Having some defaults for these will definitely be useful. And some way of allowing users to adjust the baseline to add/remove devices would also be useful.

DiamondJoseph, 2024-11-27:
This could be something that we add to the device_factory decorator: blueapi constructs the set of baseline devices and injects them as a default argument with the resolution of #874. 

Optionally some endpoint from blueapi to edit what the baseline is, but I would prefer it is defined only in the beamline module and invariant, to make debugging it simpler.

stan-dot, 2024-11-27:
yeah definitely this needs to wait for #874

I assume some top level declaration in the beamlines/[beamline] namespace or actually a proper BeamlineDefinition class where the devices are inside and various beamline metadata is hosted as encapsulated properties

as this part will get only bigger

[image]