DiamondLightSource/dodal #775: Remove alias for DetectorParams.override_run_number

DiamondJoseph, 2024-09-05:
To maintain compatibility with GDA and Hyperion,DetectorParams.override_run_number has an alias of `run_number`.

When changes between those systems can be orchestrated, a small nicety would be to have this field named `override_run_number` everywhere, to make it clear what the purpose of the field is and why it is optional. 

## Acceptance Criteria
* Specific criteria that will be used to judge if the issue is fixed

DominicOram, 2024-09-05:
It may also be that setting `run_number` externally is not required. We should think about and discuss this with scientists though.

DiamondJoseph, 2024-09-05:
The added test actually revealed [this inconsistency in Pydantic2 alias handling](https://github.com/pydantic/pydantic/issues/8379), so if we ever serialise the DetectorParams, and require deserialising where run_number is used, we have to `det_params.model_dump(by_alias=True)`. If we only ever deserialise, we are consistent with prior behaviour.