DiamondLightSource/dodal #780: Expose help and docs from device factory functions.

DiamondJoseph, 2024-09-09:
After #597 but also relevant for the current state:

Device classes have docstrings about their capabilities/use/expected behaviour. Specific device factory functions may also have docstrings about their expected behaviour, other usage information. Some combination of these should be presented to the user when they use the python builtin `help` function on the device_factory_function. 

## Acceptance Criteria
* After importing `ixx.device`, `help(device)` gives useful information about using the device, which may include
* a description of any parameters that are added by our decorator and either their default or how to find their default
* (if provided) docstring of any parameters that may be passed into the decorated function
* (if provided) docstring of the device class that is being returned
* (if provided) docstring of the device instance

DiamondJoseph, 2024-09-09:
Note that the `help` method uses the docstrings generated via pydoc, which [may require that we are using numpydoc](https://stackoverflow.com/a/21948562) style docstrings. @coretl has opinions about the verbosity.

coretl, 2024-09-09:
> Note that the `help` method uses the docstrings generated via pydoc, which [may require that we are using numpydoc](https://stackoverflow.com/a/21948562) style docstrings. @coretl has opinions about the verbosity.

Really? That SO post seems to be about Spyder. `help` seems to just print what you wrote:

```python
class A:
    def google_meth(self, a: int):
        """A google style doc method.
        
        Args:
            a: The parameter
        """
    def numpy_meth(self, b: str):
        """A numpy style doc method

        Parameters
        ----------
        b
            The parameter
        """
help(A)
```
```
class A(builtins.object)
 |  Methods defined here:
 |  
 |  google_meth(self, a: int)
 |      A google style doc method.
 |      
 |      Args:
 |          a: The parameter
 |  
 |  numpy_meth(self, b: str)
 |      A numpy style doc method
 |      
 |      Parameters
 |      ----------
 |      b
 |          The parameter
```