DiamondLightSource/dodal #782: Read aperture/scatterguard hardware on read

DominicOram, 2024-09-12:
Following https://github.com/DiamondLightSource/dodal/pull/769 a side effect of this is that we're no longer reading the state of the hardware whenever we read `selected_aperture`. This means that the device itself is holding on to the state of the beamline based on what state it last moved to. This means that the state could be wrong if:
* The device has just been initialised
* The hardware has been moved via EPICS/manually
* There was a failure in moving the device

## Acceptance Criteria
* Hardware is read when the device is read
* The device holds no assumptions about the state of the hardware

DiamondJoseph, 2024-09-13:
`selected_aperture` and `radius` are both going to be cached in the soft_signal state, so if you're modifying the handling of one should adjust the other too. Mea culpa.

DominicOram, 2024-09-13:
I wonder if we should try and write a general rule in the docs of "devices should avoid holding state and instead read the state of the hardware when read"? Thoughts @callumforrester, @dperl-dls?

DiamondJoseph, 2024-09-13:
Would be nice if there was some utility for making a soft_signal backed by some function that may read other parts of device state

PR #789 (2024-09-27): Read hardware on every read of the aperture scatterguard 
Fixes #782

### Instructions to reviewer on how to test:
1. Confirm tests now all pass
2. Confirm docs make sense

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant  [... cut]
closed 2024-09-27 by DominicOram