DiamondLightSource/dodal #840: Add linter to enforce import rules

DominicOram, 2024-10-14:
In `mx-bluesky` we have enforced some import rules with a linter https://github.com/DiamondLightSource/mx-bluesky/pull/568. We coudl do the same with `dodal`. The rules will probably be at least:
* `common` doesn't import from `devices`
* `devices.ixx` doesn't import from `devices.iyy`

## Acceptance Criteria
* The idea of linting is discussed
* We implement rules if we agree on some
* Accompanying docs are written if the rules are not clear

callumforrester, 2024-10-14:
How about "nothing within dodal imports from `beamlines`, excluding tests"? `beamlines` is essentially just configuration and I don't think anything should depend on it.

DiamondJoseph, 2024-10-14:
Having just stumbled into making a circular dependency: love it.

rtuck99, 2024-10-15:
Also `plans` can depend on `devices` but not the other way around

stan-dot, 2024-10-15:
> How about "nothing within dodal imports from beamlines, excluding tests"? beamlines is essentially just configuration and I don't think anything should depend on it.

what if we make 'beamlines' just YAML? :eyes:

callumforrester, 2024-10-15:
That possibility was discussed at the bluesky workshop at NOBUGS. I think if we went that route we would be trying to find a multi-facility solution for loading devices. Currently everyone has rolled their own using either databases, config files, or code-as-config (like dodal).

DominicOram, 2024-10-15:
Given the feedback on this I think we can say that it's been agreed upon that we do it. Updating acceptance criteria and pulling into next hyperion sprint

PR #988 (2025-01-15): Add import linting
Fixes #840 

### Instructions to reviewer on how to test:
1. Confirm correct import directionality

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant beamline(s) been run via `dodal connect  [... cut]
closed 2025-01-15 by shihab-dls