DiamondLightSource/dodal #845: Deprecate `inject` method

DiamondJoseph, 2024-10-17:
`inject` only worked by coincidence of Pydantic coercion: now that we can have device singletons created without connect being called until appropriate, we can use the device instances as the default arguments to plans.

## Acceptance Criteria
* inject has been marked as deprecated
* plans and stubs that currently make use of it in dodal have been replaced by a call to the device factory of the appropriate name called with `(connect_immediately)=False`
* documentation has been updated to reflect this new recommended way of injecting default arguments into plans
* plans in our plan repositories (mx-bluesky, i22-bluesky, etc.) have a pathway (issues or PRs) for migrating to the new method

closed 2024-10-18 by callumforrester

dperl-dls, 2024-10-21:
> `inject` only worked by coincidence of Pydantic coercion

Could you add some more details about this?

DiamondJoseph, 2024-10-21:
> Could you add some more details about this?

`inject` was just lying to the type checking: blueapi maintains a dict of `TypeReference[T]` and when you register a plan knows that to convert from a `str` to a `T` involves fetching it from its internal context. It was found when we were looking at default args that this worked, and was intended as a temporary solution.

I'll be honest, I don't entirely know where in Pydantic the default argument was being converted to `T`: whether in creating the Model class for the PlanFunction or creating the instance of the model for the plan.

I've just had a thought that if it was the former (during creation of the model class for the plan function) that if you changed your blueapi config to load the plans prior to the devices it might fail in the way we've seen...

Using a `@device_factory` decorated non-connected device avoids lying to the type checking, the `TypeReference` then is only relevant to blueapi (where the context is and needing to handle `str` arguments from REST) and the implied but not guaranteed mapping of device.name to device instance is removed in favour of the factory function with the correct args.

PR #854 (2024-10-18): Add deprecation warning to inject
Fixes #845 
Supersedes #839 

### Instructions to reviewer on how to test:
1. Confirm tests pass
2. Optional: write a plan that uses `inject` and confirm it raises a warning when run.

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans a [... cut]