DiamondLightSource/dodal #850: I22 connection tests did not fail when they weren't mocking

callumforrester, 2024-10-17:
#847 appears to have been caused by #841 not passing `mock=True` to the `DeviceInitializationController`, that should have caused a timeout failure when the CI tried to connect to PVs that don't exist on its network, instead the tests passed after a very long time, suggesting that something was timing out but not failing the test, perhaps an exception was being swallowed.

This issue is to investigate and prevent recurrence

## Acceptance Criteria
* The cause of the non-failure is known
* A regression test/tests are in place to prevent recurrence

DiamondJoseph, 2024-10-18:
make_all_devices returns the connected devices and also collects all of the exceptions that were raised.
As discussed, the tests should have accounted for checking the collected exceptions.

rtuck99, 2024-10-25:
This is fixed in #858

PR #868 (2024-10-25): 850 fix for exceptions being ignored in beamline unit tests
Fixes #850

### Instructions to reviewer on how to test:
1. Unit tests all pass

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant beamline(s) been r [... cut]
closed 2024-10-25 by rtuck99