DiamondLightSource/dodal #864: Dodal connect on i22 is not actually connecting

DominicOram, 2024-10-24:
If I run `dodal connect i22` I get all of the beamline connected suspiciously quickly. It turns out that if I edit one of the PV prefixes in `i22.py`to be incorrect then `dodal connect i22` will still pass.

## Acceptance Criteria
* If I deliberately break one of the PV prefixes in i22.py then `dodal connect i22` actually fails

DominicOram, 2024-10-24:
@callumforrester and @DiamondJoseph am I right that this is different to https://github.com/DiamondLightSource/dodal/issues/850?

DiamondJoseph, 2024-10-25:
This is different than 850: 850 is that when make_all_devices is called, any exceptions in the `device_instantiation` way of creating and connected devices are swallowed and do not cause test failures.
This is that make_all_devices doesn't ever call connect on the devices given by `@device_factory` on the returned devices after they are made- make_all_devices may catch exceptions in instantiating the device, but not in the connect which is what's more useful...

callumforrester, 2024-10-25:
Additional acceptance criteria: If we could have a regression unit test that would be great

DiamondJoseph, 2024-10-25:
#865 which is intended to fix this issue (#864) is also running into the same broken connects that were hidden by #850 so they are very much related

DominicOram, 2024-10-28:
With https://github.com/DiamondLightSource/dodal/pull/869 reverted I don't believe this issue is on main now. However, we will need to address it before reinstating the device factory

callumforrester, 2024-11-11:
The cause of this issue that `dodal connect <beamline>` does not attempt to connect the devices, but expects `device_instantiation` to do it, which is does as long as `wait_for_connection=True` is passed. In practice this is true often enough that no one has really noticed, but it is technically still a bug in main because `dodal connect` does not work for lazy devices. It is just made worse in the world of `DeviceInitializationController` because all devices are lazy by default there.

So the fix is to make sure the dodal CLI actually calls `ensure_connected` on the devices it connects to.

closed 2024-11-19 by callumforrester