DiamondLightSource/dodal #926: BeamlinePrefix does not support branch beamlines (e.g. i13-1 does not resolve to `BL13J`)

DiamondJoseph, 2024-11-26:
get_beamline_name returns what is set in the environment variable BEAMLINE, or a default string. 
The return of get_beamline_name is (often) used to create the BeamlinePrefix, which is used by both device_instantiation and device_factory to create the device prefix.

On e.g. i13-1-ws001, BEAMLINE=i13-1, and BeamlinePrefix("i13-1").beamline_prefix = "BL20I" however devices on i13-1 are actually prefixed `BL13J`.

Is using the environment variable to connect, other than in the cases of a simulated twin, a common behaviour that we need to support? This is done by the training_rigs, which share a dodal beamline module, and so must be distinguished.

## Acceptance Criteria
* Specific criteria that will be used to judge if the issue is fixed

noemifrisina, 2024-11-27:
IIRC, in a case such as `i13-1`, specifying the suffix in BeamlinePrefix should do the trick - even if it's not a very nice solution (and we might want something else in the long run):

```python
from dodal.utils import BeamlinePrefix

bl = BeamlinePrefix("i13-1", suffix="J")
```
This should return `bl.beamline_prefix = "BL13J"

Btw, just to make things more interesting, I'll note that the beamline prefix for both i19-1 and i19-2 is `BL19I` but that might be a single edge case.

DominicOram, 2024-11-27:
FYI, this is a duplicate of https://github.com/DiamondLightSource/dodal/issues/760. I will close the older one

DominicOram, 2025-07-31:
I think `bl = BeamlinePrefix("i13-1", suffix="J")` is a good enough solution to this for now and seems to be working on a few beamlines

closed 2025-07-31 by DominicOram