DiamondLightSource/dodal #927: Create a filter wheel attenuator device

noemifrisina, 2024-11-27:
Some beamlines, eg I24 and soon I19, have filter wheels on their attenuators, for which there is not yet a device

## Acceptance Criteria
* There is a fairly general filter wheel device with common features that beamline-specific attenuators can use

DiamondJoseph, 2024-11-27:
Is the implementation anything similar to p99's [FilterMotor](https://github.com/DiamondLightSource/dodal/blob/cb7717c2e1f73dc696227e98cb44acb16c4fd7f8/src/dodal/devices/p99/sample_stage.py#L33), with an enum of possible positions?

olliesilvester, 2024-12-19:
> Is the implementation anything similar to p99's [FilterMotor](https://github.com/DiamondLightSource/dodal/blob/cb7717c2e1f73dc696227e98cb44acb16c4fd7f8/src/dodal/devices/p99/sample_stage.py#L33), with an enum of possible positions?

Yes, same thing here - but the enum options are different