DiamondLightSource/dodal #933: Move blueapi documentation on device best practices into dodal

callumforrester, 2024-11-28:
Based on [this discussion](https://github.com/DiamondLightSource/blueapi/pull/674#discussion_r1804707810) we should move all information about how to create devices and configure a beamline into the dodal docs and only link from the blueapi docs.

## Acceptance Criteria
* The information in the page linked above is in the dodal docs
* The [create a beamline](https://diamondlightsource.github.io/dodal/main/how-to/create-beamline.html) and [make a device](https://diamondlightsource.github.io/dodal/main/how-to/make-new-ophyd-async-device.html) how-tos do not conflict with this information, are up-to-date and complete

noemifrisina, 2025-08-27:
This looks like it has been mostly done over the past few months.

callumforrester, 2025-08-28:
Agreed, shall we close it?

closed 2025-08-29 by callumforrester