DiamondLightSource/dodal #942: Momentary signals mechanism

barnettwilliam, 2024-12-05:
Is there a mechanism for momentary signals in dodal? 

Currently synoptic buttons that send a value when pressed and another value when released to momentarily set the value of an ioc's PV. For example on the pressure cell valve OPEN_SEQ  pvs e.g. BL38P-EA-HPXC-01:V1:OPENSEQ. For the pressure jump cell case a delay using `asyncio.sleep` was used but this kind of signal could be standardised in dodal.

Orginally came up in -  https://github.com/DiamondLightSource/dodal/pull/812#discussion_r1787716082

olliesilvester, 2024-12-05:
I haven't seen a standardised way for this, but I think using `asyncio.sleep` like you did in https://github.com/DiamondLightSource/dodal/commit/c46fb8e3992044181b8f7dd038e140ca59791d5a is fine. 

I guess we could add some kind of `set_pulsed_signal` helper function, but the logic is quite simple and short, so I would argue there's no need

DominicOram, 2025-01-21:
I think for most of the `PROC` PVs we've come across in MX we just send a 1 and we don't need to set it back to 0 as the IOC will do that itself. Is this definitely needed in this case? What does the EDM screen do when it's triggered, a similarly small sleep? I would suggest this might be a good place to standardise more on the EPICS side so that it's not DAQ's responsibility to reset these signal's to low.