DiamondLightSource/dodal #943: Fix flakey bart robot unit test

olliesilvester, 2024-12-05:
One of our tests for the bart robot occasionally fails. We should fix this

`tests/devices/unit_tests/test_bart_robot.py::test_given_program_not_running_and_pin_unmounts_then_mounts_when_load_pin_then_pin_loaded - 
AssertionError: Expected 'trigger' to have been called once. Called 0 times.`

Or see https://github.com/DiamondLightSource/dodal/actions/runs/12165707719/job/33930319285

## Acceptance Criteria
* Test made more robust

paulorochaoliveira, 2024-12-05:
FYI, I'm taking a look at this flaky test right now.

paulorochaoliveira, 2024-12-05:
It seems like what makes this test flaky is the sleep time sometimes being to short. Increasing the sleep time would most likely fix the issue.
Another more robust solution is to replace arbitrary sleeps with an event-driven approach. For example, use asyncio.Event to synchronize the test with the actual behavior.
What do you think @olliesilvester ?

olliesilvester, 2024-12-05:
I think increasing the sleep is good enough for this issue, however I like the idea of using `asyncio.Event` instead. I will make an issue to replace all of our arbitrary sleeps in tests instead of this! Thanks!

paulorochaoliveira, 2024-12-05:
Should I open a PR to at least close this issue?

olliesilvester, 2024-12-05:
Yes, go ahead, thank you

DominicOram, 2025-01-21:
I think we need to properly mock it and rethink how the test works rather than just change the sleeps. e.g. we could patch sleep like in https://github.com/DiamondLightSource/dodal/blob/f93183fb50d3012c2c4074508e09792c43660fbe/tests/devices/unit_tests/test_thawer.py#L29 so that we can be very specific on when things should be happening

PR #1392 (2025-07-29): Remove sleeps from bart robot tests
Fixes #943 (hopefully)

These tests relied on sleeps to order when to set signals. I've changed it so that the signals are set based on specific logging messages. Not sure if it's the _best_ fix but it should work for now

### Instructions to reviewer on how to test:
1. Confirm tests are still sensible
2. Confirm flakiness is resolved

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference [... cut]
closed 2025-07-29 by DominicOram