DiamondLightSource/dodal #954: Create pytest marker to easily run local unit tests

olliesilvester, 2024-12-10:
To run unit tests locally, we now need to type `pytest -m "not (s03 or adim)"`, and the list of things which we are skipping is going to grow. We should add a `unit tests only` flag which does all the 'nots' for us

## Acceptance Criteria
* Add pytest flag to run local unit tests

NINAD-KHADSE, 2024-12-11:
Can you assign this to me

olliesilvester, 2024-12-12:
Sure, I think we want to add something in our `pytest_runtest_setup` so that running `pytest -m unittests` does the same as `pytest -m "not (s03 or adim)`. https://docs.pytest.org/en/stable/example/markers.html might help

olliesilvester, 2025-01-14:
`tox -e tests` gives us this functionality given that `[tool.tox]` is configured correctly in `pyproject.toml`

closed 2025-01-14 by olliesilvester