DiamondLightSource/dodal #1001: Eiger: Create a gantt chart of the arming process

DominicOram, 2025-01-23:
To be able to optimise the arming of the Eiger, and to write the FastCS Eiger we need to understand the order of operations better. To do this we should make a diagram (something like a gantt chart) of what operations depend on others and how long they each take.

## Acceptance Criteria
* There is a diagram of what operations rely on what others in the eiger and how long they take

DominicOram, 2025-01-23:
Based on the code the current behaviour is:

```mermaid
gantt
    title Eiger arming with timing in seconds
    dateFormat ss
    axisFormat %S
    odin.stop                                                       : a1,  0, 1s
    change_dev_shm                                         : a2, after a1, 1s
    set_detector_threshold                                 : a3, after a2, 1s
    set_cam_pvs                                                : a4, after a3, 1s
    set_odin_number_of_frame_chunks            : a5, after a4, 1s
    set_odin_pvs                                                : a6, after a5, 1s
    set_mx_settings_pvs                                   : a7, after a6, 1s
    set_num_triggers_and_captures                 : a8, after a7, 1s
    set_mx_settings_pvs                                   : a9, after a8, 1s
    wait_for_not_stale                                       : a10, after a9, 1s
    _wait_for_odin_status                                 : a11, after a10, 1s
    set_acquire                                                 : a12, after a11, 1s
   _wait_fan_ready                                          : a13, after a12, 1s
```

shihab-dls, 2025-01-24:
I've just updated the Gantt chart; these are averaged over 4 scans (with rotation too, so 8 arming chains in total)

DominicOram, 2025-01-24:
`set_mx_settings_pvs` is odd here, it should just be setting a few PVs. Would be worth digging into it and working out which of those sets is slow

shihab-dls, 2025-01-24:
Ah, I hadn't noticed enable_roi_mode due to its conditional execution, so I agree this is likely where the final `1.069s` are. Will add logging there, and more granular logging on `set_mx_settings_pvs`

DominicOram, 2025-01-24:
For interest @GDYendell

DominicOram, 2025-01-31:
The disarm is now on the critical path so would be good to profile that too.

DominicOram, 2025-01-31:
For disarm in the case where we're not free running and already disabled ROI (i.e. a rotation scan):

```
Waiting on filewriter to finish - 1.76s
Disarming det - 0.073
Waiting on odin ok - 0.007
Disable ROI - 0.004
```

For disarm when we're free running and not ROI disabled (i.e. XRC):
```
Waiting on final frames - 0.444
Stopping odin - 1.12
Waiting on filewriter to finish - 0.001
Disarming det - 0.038
Waiting on odin ok - 0.005
Disable ROI - 2.887
```

DominicOram, 2025-01-31:
See https://jira.diamond.ac.uk/browse/I03-637 for some analysis that has been done before