DiamondLightSource/dodal #1010: Make CONTRIBUTING.md

stan-dot, 2025-01-24:
From discussion with @DiamondJoseph ; there are non-trivial acceptance criteria to PRs that could be more explicit. 

- does device connect - on one beamline, or all?
- when do we allow 'skip_device' in a PR?
should PR code correspond to the beamline reality at merge OR to the model of beamline reality and the exact details could be worked in a separate PR?

stan-dot, 2025-01-24:
current 'contributing' documentation is not as plain to see and does not cover those scenarios
https://diamondlightsource.github.io/dodal/main/how-to/contribute.html

callumforrester, 2025-01-24:
While contributing guidelines are useful I think we should be careful.

Devices may fail to connect due to temporary beamline network issues, shutdowns, power cuts, or just being occasional-use. None of these should block a Github PR from being merged. Between these I do not think there will ever be a single instant where every device on device beamline connects, so there's little point in requiring that.

>     beamline reality at merge OR to the model of beamline reality and the exact details could be worked in a separate PR?
This one harder but personally I'm against formalizing this too. @stan-dot  I think the docs you actually want are [here](https://diamondlightsource.github.io/dodal/main/explanations/reviews.html). We are trialling the approved reviewer system in the hope that these kinds of decisions will spread culturally rather than formally.  That means the people in the review team have a lot of weight in the decision. Personally I would bias myself towards beamline reality at merge.

DominicOram, 2025-05-20:
Not sure if it fits here but there are some standards on workflows that I believe that we should document:
* All major PRs (anything more than typos/small refactors) should have associated issues describing what they are trying to solve, who is requesting it, what the acceptance criteria are
* All todos in the code should be linked to such issues giving detail on what needs to be done

DominicOram, 2025-05-20:
We also already have https://diamondlightsource.github.io/dodal/main/how-to/contribute.html that may cover some of the same things as a `CONTRIBUTING.md`