DiamondLightSource/dodal #1022: Flaky test in test_utils

stan-dot, 2025-01-27:
FAILED tests/devices/unit_tests/test_utils.py::test_wrap_function_callback_errors_on_wrong_return_type - AssertionError: assert 'does not return a Status' in 'WARNING  ophyd:status.py:254 Status(obj=None, done=False, success=False) has timed out\n'
 +  where 'WARNING  ophyd:status.py:254 Status(obj=None, done=False, success=False) has timed out\n' = <_pytest.logging.LogCaptureFixture object at 0x7f687030f920>.text

https://github.com/DiamondLightSource/dodal/actions/runs/12991801896/job/36230206846?pr=722

stan-dot, 2025-01-27:
the test is successful locally though

stan-dot, 2025-01-27:
`'ERROR    Dodal:epics_util.py:73 wrap_func attempted to wrap <MagicMock id=\'139693409217488\'> when it does not return a Status\nERROR    ophyd:status.py:289 An error was raised on a background thread while running the callback functools.partial(<function run_functions_without_blocking.<locals>.wrap_func at 0x7f0ccfbf2520>, current_func=<MagicMock id=\'139693409217488\'>, next_func=<function run_functions_without_blocking.<locals>.closing_func at 0x7f0ccfbf27a0>)(Status(obj=Dodal test utils - get good status, done=True, success=True)).\nTraceback (most recent call last):\n  File "/venv/lib/python3.11/site-packages/ophyd/status.py", line 287, in _run_callbacks\n    cb(self)\n  File "/workspaces/dodal/src/dodal/devices/util/epics_util.py", line 76, in wrap_func\n    raise ValueError(f"{current_func} does not return a Status")\nValueError: <MagicMock id=\'139693409217488\'> does not return a Status\nWARNING  ophyd:status.py:254 Status(obj=None, done=False, success=False) has timed out\n'`

the log is like above. It arguably contains the requested string

stan-dot, 2025-01-27:
locally was running: 
```ophyd==1.10.0
ophyd-async==0.9.0a1

```

stan-dot, 2025-01-27:
running again, trying debugging logging

stan-dot, 2025-01-27:
now it worked. looking for reasons of non-determinism
```tests: commands succeeded
  congratulations :)```

stan-dot, 2025-01-27:
@DominicOram I took a quick look into this - does the draft branch make any sense?

DominicOram, 2025-01-27:
> [@DominicOram](https://github.com/DominicOram) I took a quick look into this - does the draft branch make any sense?

No, I don't think that will work, you need to patch the full path to where it's imported (https://stackoverflow.com/questions/11351382/mock-patching-from-import-statement-in-python).

This test is pretty horrible and niche, if it's flakiness is holding you up feel free to comment it out and link to this issue

stan-dot, 2025-01-27:
not really it's not holding me up[