DiamondLightSource/dodal #1042: Helper function to list the PVs of a device

DominicOram, 2025-02-06:
Often it is useful to find the PV that a particular signal is talking to whilst developing code e.g.
* To be able to search for it in the archiver
* To look for the db logic behind it
* To do a `cainfo` on it

However, it's quite hard to find this from the code given that it may go through a few levels of adding prefixes etc. Thus it would be useful to be able to do something like `dodal connect i10 --list_pvs` and get a list of all signals and what their PVs are.

## Acceptance Criteria
* There is an easy way to find what PVs are in use when developing code

DominicOram, 2025-02-06:
@DiamondLightSource/bluesky-reviewers - I would appreciate any other ideas here? In an ideal world I would be able to find the PV for a particular signal as easily as possible. Listing them all and searching is easier than what we currently have but if we have an easier way that would be great

noemifrisina, 2025-02-06:
I think that would be a really useful feature. Only comment I have is that I wouldn't actually want the devices to connect every time, and I don't think that's needed to get a PV list, making them should be enough?
Reason for that is that we often may be doing development during shutdown/machine day, when the hardware on the beamline could be offline for any reason. This would cause `dodal connect` to fail, making the output difficult to read...

olliesilvester, 2025-02-11:
This would be useful, I agree that they shouldn't actually try to connect though. The `--list_pvs` flag can change the behaviour to whatever we want though, so we can just make it not actually connect.

My suggestion would be to be able to use it like:
`dodal get_pvs {beamline}` to list all PV's from that beamline module
`dodal get_pvs {beamline}.{device}` to list all PV's from that beamline's device
`dodal get_pvs {beamline}.{device}.{attribute}` to list all PV's from that part of a beamline's device

And we should be able to get more and more specific until the signal level

olliesilvester, 2025-02-12:
> Just a note that for PandAs that use information from their PVI signal to build up the model of their blocks, so it wouldn't work for those devices or any future devices that use PVI. 
> ```

Good point, these are the devices which are especially annoying to find the PV names for too :(

DominicOram, 2025-02-12:
> Just a note that for PandAs that use information from their PVI signal to build up the model of their blocks, so it wouldn't work for those devices or any future devices that use PVI. Otherwise seems very useful, something structured rather than a flat list presumably?
> 
> e.g.
> 
> ```
> foo
>   foo:bar
>     foo:bar.VAL
>   foo:baz
> ...
> ```

Not for a first pass of this but presumably a future version could connect, interrogate the PVI structure and then report on it.

callumforrester, 2025-02-12:
I like this idea, is it worth raising this issue against ophyd-async?

callumforrester, 2025-08-01:
Done in https://github.com/bluesky/ophyd-async/pull/917

closed 2025-08-01 by callumforrester