DiamondLightSource/dodal #1076: Create a device catalogue

olliesilvester, 2025-02-26:
As more groups across DLS are starting to use dodal, it will be very useful to have an organised list of all the common devices, along with a simple description of how that device is used in a beamline (from a hardware perspective).

Once a few of these have been written, I think we should start to ask that PR's adding new devices must also come with a new catalogue entry (open to discussion on this part)

We should create sub-issues to add individual devices to the catalogue

## Acceptance Criteria
- Device catalogue created, containing entries for the most commonly-used devices
- Document that when adding a new device, you should also add an entry to the device catalogue

DominicOram, 2025-02-26:
I think we should be able to ingest the docstrings of the devices and put them into the developers portal via backstage

callumforrester, 2025-03-13:
How different would we want it to be to the reference docs ([beamlines](https://diamondlightsource.github.io/dodal/main/reference/generated/dodal.beamlines.html) and [devices](https://diamondlightsource.github.io/dodal/main/reference/generated/dodal.devices.html))?

oliwenmandiamond, 2025-08-27:
I think the issue with using the generated docs there's no "common" devices. I think a simpler solution to this could be to create dodal/devices/common/ and move anything common to there.

That we generated docs would create a directory of docs for common devices and any one browsing the repo can clearly see devices marked common. This would also help tidy up the device structure a bit IMO

callumforrester, 2025-08-27:
My $0.02:

- I'm generally opposed to adding yet more structure to dodal, it's already quite complicated and it can hard to find things
- My assumption was that anything not inside a beamline directory is "common" by default, so there's no need to add more heirachy to explicitly note this
- I think this issue actually has potential to be a very good solution to the hierarchy problem, if we can end up with a flat list of devices with an auto-generated catalogue that allows us to query by description, beamlines in-use, type and various other parameters, that would be much better and more usable than trying to come up with a hierarchical structure that makes sense to newcomers, experienced engineers, people from different science groups etc.