DiamondLightSource/dodal #1092: Remove undulator_dcm

olliesilvester, 2025-03-05:
The `UndulatorDCM` composite device was created so you can change beamline energy in Bluesky by acting on a single device, with the logic abstracted away. However, this is causing some issues

- Setting energy sometimes requires more devices than just these two, eg `MirrorVoltages`
- Typing gets messy if a function which expects there to an `Undulator` instead sees an `UndulatorDCM`
- Logic acting over multiple devices should (arguably) belong in a Bluesky plan

I propose we remove this composite device, and its the logic can be moved into Hyperion's `set_energy_plan`. As far as I'm aware, it isn't being used anywhere else. @DiamondLightSource/bluesky-reviewers @stan-dot please let me know if this device is being used anywhere else!

At a later date, a generic `change_energy` plan can be made in dodal

## Acceptance Criteria
- UndulatorDCM removed
- The logic in the above is moved to [Hyperion's set_energy_plan](https://github.com/DiamondLightSource/mx-bluesky/blob/main/src/mx_bluesky/hyperion/experiment_plans/set_energy_plan.py)

DiamondJoseph, 2025-05-21:
I assume @stan-dot's thumbs up means the UndulatorDCM isn't being used anywhere else. I agree that this logic belongs in a plan rather than in a device.

stan-dot, 2025-05-21:
yes, I see I could have been more explicit about this being ok