DiamondLightSource/dodal #1105: I24 serial: add a monitor for the frame counter to PMAC ProgramRunner

noemifrisina, 2025-03-11:
Once #1104 is done, we need to add a monitor for the  frame counter PV on the PMAC. This monitor should determine whether the collection is done or if the pmac has stopped prematurely.

eg.
If the expected number of frames is 25600, the ProgramRunner should complete once this number is reached or timeout if the frame counter value has not reached the value but hasn't changed in 10-30s.

## Acceptance Criteria
* The PMAC ProgramRunner has a monitor that checks the fram counter for either collection done or timeout.

DominicOram, 2025-03-12:
To do this take a look at https://blueskyproject.io/ophyd-async/main/_api/ophyd_async/ophyd_async.core.html#ophyd_async.core.observe_value. I think the timeout on that is almost exactly what we want.

noemifrisina, 2025-04-14:
Small addendum. We just saw an issue on the beamline where the motors got stuck and went in an error state but the bluesky plan kept waiting for the collection to continue. This is because apparently in case of an error on a motor the motion program doesn't error but keeps running - meaning the status PV we're currently monitoring never goes back to 0. 

Closing this issue should also help fix the problem: if the motor doesn't move the counter PV will not update, timing out in a few seconds. (Other solution would be to monitor and poll the motors state over the run, but hopefully this is overkill)

DominicOram, 2025-05-06:
Rather than wait for the full number of frames, as this needs the number of frames to be passed into the runner we could just wait for the status to be don but whilst waiting we timeout if the counter has stopped increasing for 30s

PR #1216 (2025-05-09): 1105 add monitor for frame counter
Fixes #1105 and completes #900

### Instructions to reviewer on how to test:
1. Confirm tests still pass

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant beamline(s) been r [... cut]
closed 2025-05-09 by adaudon