DiamondLightSource/dodal #1120: Add electron analysers as devices with their region data

oliwenmandiamond, 2025-03-17:
For surface beamlines, they use electron analysers. These are detectors that usually read in a file that define a list of region settings. They read in these settings, configure the detector, and then perform data collection. 

To do this in bluesky, we need to create the initial base analyser class along with the region classes.

Villtord, 2025-03-17:
This change is mixed up a bit, I would perhaps brake this into Regions and Sequences purely and PR that one first, and add another one as base analyser and specific implementations. For the latter I would also introduce a controller class that will take all the epics communications, similar to what we have in Java now. I would argue we need to work in smaller changes as otherwise it is really hard to contribute in synch.

oliwenmandiamond, 2025-03-17:
Make a good point, I can take out analyser part out and that way everything else is 100% test covered too. 

The controller class is the BaseAnalyser Device class and then the specialised "controller" classes inherit from it. These control the ability to set the PV's so configure with the region settings.