DiamondLightSource/dodal #1129: Make a default PATH_PROVIDER to write to tmp if one if not provided

abbiemery, 2025-03-18:
With the development of using numtracker for a path provider we would want a default path provider to exist for if people don't set one. This is with the idea that blueapi will control the path provider going forward. We need to discuss how we want to go forward with this.

```
PATH_PROVIDER: PathProvider = StaticPathProvider(UUIDFilenameProvider(), "/tmp")
```

DiamondJoseph, 2025-08-11:
Currently the only other instrument consistently set up to use the path provider with NumTracker integration is the adsimulator, which has a [lengthy comment about setting up](https://github.com/DiamondLightSource/dodal/blob/440b816b1bbc4a94fa5fa95d1f69f9081ec18f35/src/dodal/beamlines/adsim.py#L19) for e.g. running in a terminal. It's unlikely that `dodal connect adsim` is going to be run very often.

p46-9 will be switched over to using it imminently, and are more likely to need `dodal connect p4x`. @abbiemery has a ticket this sprint to move another beamline over to using it.

`dodal connect` could set the path provider to `StaticPathProvider(UUIDFilenameProvider(), "/tmp")`, otherwise requiring explicitly setting it- at the top of the beamline module (as done currently), in the terminal [trying to connect to a device before it is set should have an obvious error message] or in blueapi when prior to creating devices

DiamondJoseph, 2025-08-11:
```python
def get_path_provider() -> PathProvider:
    if PATH_PROVIDER is None:
        raise ValueError("""
            A device requires configuration for writing to the filesystem that has not been set.
            Configure this by setting a PathProvider:

            from pathlib import Path
            from ophyd_async.core import StaticPathProvider, UUIDFilenameProvider
            from dodal.common.beamlines.beamline_utils import set_path_provider

            set_path_provider(
                StaticPathProvider(
                    UUIDFilenameProvider(),
                    "/tmp", # The directory for `det` to write to- may be mounted as a volume
                )
            )"""

    return PATH_PROVIDER
```

Doesn't require any extra looking up docs, just raises the exception when you try and

```python
$ from dodal.beamlines.adsim import det
$ d = det()
# or
$ from dodal.beamlines import adsim
$ from dodal.common.utils import make_all_devices
$ make_all_devices(adsim)
```

callumforrester, 2025-08-11:
I feel at this point we are just differing on fundamental philosophy on how systems should work: "Works sensibly out of box, consult the instructions for more complex usage" vs. "Read the instructions first you lazy so and so". @olliesilvester I think we need a tiebreaker, I will do whatever the consensus says.

DiamondJoseph, 2025-08-11:
RE: discussion IRL: Callum thinks that he, Gary and possibly other controls engineers just want to know *that data is written*, and don't require that it is written to a commissioning directory or visible, so having the default path provider that writes to /tmp won't trip people up as I think it might do if it is set vs. requiring someone to make a decision on where files are written. 

There may be something to think around consistent behaviour of IOCs and how often people will use a terminal vs. use blueapi but I'm going to cede.

abbiemery, 2025-08-11:
With the way it is now, as `PATH_PROVIDER` is a global, if you call `get_path_provider` it will `ValueError` if it hasn't been set yeat.

abbiemery, 2025-08-11:
For what it is worth I do think a default provider to `tmp` is a good idea, but we implement this is a different story. 

This is the nth time we've hit our head against this, can we please stop using global state for this? If the problem is continually difficult to solve, then we're trying to solve it in the wrong way.

abbiemery, 2025-08-11:
Blueapi shouldn't depend on dodal, and we shouldn't be using dodal module globals like this.