DiamondLightSource/dodal #1136: Add HDF suffix in adsim detector for simulated beamline

Villtord, 2025-03-24:
It looks like currently dodal/beamlines/adsim.py uses "HDF5" suffix for detector writer, however the link to a simulation beamline in file comments sets up IOC that uses "HDF" suffix. Following this comments lead to a connection failed error for hdf PVs.

## Acceptance Criteria
* Change does not disturbs other possible simulated devices
* Change works for simulated beamline https://epics-containers.github.io/main/tutorials/launch_example.html

DominicOram, 2025-03-24:
Given that our beamlines all use `HDF5` is it possible to get the simulator PVs changed @gilesknap? Looks like it would just be a change in https://github.com/epics-containers/example-services/blob/b97720e0689f5459facb5c485d22f792f0fbd862/services/bl01t-di-cam-01/config/ioc.yaml#L77?

Villtord, 2025-03-24:
Yep that would be another way to do it.

gilesknap, 2025-03-24:
Sure thing. I don't think anyone else is using the simulated beamline verbatim so we can get away with changing it.

Are there other plugins that don't match? what other plugins would you like to have if any?

gilesknap, 2025-03-24:
For now, a new version of example services has :HDF5: instead of :HDF: in it's PV names.
See https://github.com/epics-containers/example-services/releases/tag/2025.3.5

I did not know you were using this for tests. Now I get to find out if anyone else is :-)

Villtord, 2025-03-25:
@gilesknap I only noticed HDF not working so far, will keep an eye on others. I am also not sure we will use it for development, but on the other hand if it is close enough to a "real beamline" then why not.

Villtord, 2025-03-25:
Going to close this issue then.

closed 2025-03-25 by Villtord