DiamondLightSource/dodal #1140: Use `ophyd_async` DerivedSignals

DominicOram, 2025-03-25:
Following https://github.com/bluesky/ophyd-async/pull/661 there is now a DerivedSignal in ophyd_async that should be able to replace the `hardware_backed_signal` we have made in `dodal`.

## Acceptance Criteria
* `hardware_backed_signal`s are replaced by `DerivedSignal`s from `ophyd_async`
* Tests still pass
* Any issue with the interface are fed back to ophyd-async

Relm-Arrowny, 2025-03-25:
Can I pick this up, I am going to have to do it for i10 id and current amplifier later.

DominicOram, 2025-03-25:
Note that you will have linting/test issues until https://github.com/DiamondLightSource/dodal/pull/1143 is sorted

Relm-Arrowny, 2025-03-27:
I have a quick try at making the change and I run into an issue, it appears they signal type is extracted from the type hint of the transformation function but at the moment it only support python build in type, if I try to us a custom type I get the following error:
`  File "/usr/local/lib/python3.11/typing.py", line 2414, in get_type_hints
    hints[name] = _eval_type(value, globalns, localns)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/typing.py", line 395, in _eval_type
    return t._evaluate(globalns, localns, recursive_guard)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/typing.py", line 905, in _evaluate
    eval(self.__forward_code__, globalns, localns),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
NameError: name 'ApertureValue' is not defined`

which is caused by [get_type_hints](https://github.com/bluesky/ophyd-async/blob/d3ea0c91793c8585db476f021b0fe2f97882d871/src/ophyd_async/core/_derived_signal.py#L40C26-L40C40) as ApertureValue is not define in _derived signal scope, I can make the device work by defining  ApertureValue within _derived_signal.py.

**To Reproduce**
Run test_aperture_scatterguard.py

**Expected behavior**
I think we expect to be able to use custom type in derived signals.

DominicOram, 2025-03-27:
I think this can currently be done with the more complete API https://blueskyproject.io/ophyd-async/main/how-to/derive-one-signal-from-others.html#multi-derived-signal. I think it would be good to have this in the simple API if we can though, a common usecase we have is converting an `enum` of positions to/from motor values. Thoughts @coretl?

PR #1148 (2025-04-25): 1140 use ophyd async derivedsignals
Fixes #1140 

### Instructions to reviewer on how to test:
1. Check the changes are sensible. 
2. Confirm test still pass.

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant [... cut]
closed 2025-04-25 by Relm-Arrowny