DiamondLightSource/dodal #1189: Move util functions out of dodal repo

callumforrester, 2025-04-23:
Separate out utilities into separate module. 

Per discussion in https://github.com/DiamondLightSource/mx-bluesky/issues/922#issuecomment-2820557556, we think dodal should be used purely for plans, devices and beamline config. Utility functions such as `make_all_devices` belong somewhere else because they're hard dependencies, not plugins.

Still to do
- Identify all functions we'd like to move
- Decide where to put the code? Blueapi, new module, somewhere else?

## Acceptance Criteria
* TBD

callumforrester, 2025-04-25:
For a good overview of util functions that are used _outside_ of dodal:

https://github.com/search?q=org%3ADiamondLightSource+%28NOT+repo%3ADiamondLightSource%2Fdodal%29++%28%22from+dodal.utils%22+OR+%22from+dodal.common%22%29&type=code

olliesilvester, 2025-05-08:
Some thoughts/opinions:

- There are a few dodal functions which are required for BlueAPI to actually startup.  These bits should just go into `BlueAPI`. For example, it looks like `make_all_devices` and the path provider bits fit this criteria
- There are some functions like `inject` which are used for example plans and in tests, but not in any central logic. We need to decide: Are these functions essential to someone using BlueAPI? If yes, then move them to BlueAPI. If no, we should change the tests and example plans so that they don't use them.

For now, I have no opinion on the dodal util functions which aren't being used by BlueAPI, like `get_beamline_name`. This hasn't caused any issues yet (for me anyway)

callumforrester, 2025-06-04:
I have a very dim memory of @coretl talking me out of requiring people to type `from blueapi import <thing>` while developing plans, but I can't recall the details

coretl, 2025-06-09:
> I have a very dim memory of [@coretl](https://github.com/coretl) talking me out of requiring people to type `from blueapi import <thing>` while developing plans, but I can't recall the details

I think this was a circular dependency issue. If you want to make a generic plan you might put it in `dodal`. If it imports `inject` from `blueapi` then `blueapi` can never import `dodal`. If that is the purpose of this PR then this is no longer a problem. As long as `device_factory` and `inject` are in the same repo this is probably fine.