DiamondLightSource/dodal #1316: Energy change plan shouldn't move the mirror + dcm motors unless necessary

rtuck99, 2025-07-01:
Since the optics motors operate in a vacuum even small changes can result in excess heat generation.
It has been found that even though the motors are set to essentially the same value this results in a rise in motor activity and temperatures
https://diamondlightsource.slack.com/archives/C048RD0KFFF/p1751299650243839

We should really only be running the change energy plan if there's a change in energy or it's the first run, alternatively this could be a controls fix.

## Acceptance Criteria
* DCM, mirror, energy changes are minimised.

DominicOram, 2025-07-01:
We should discuss with controls whether they could implement something their end that does not energise the motors if a move is requested in a deadband

DominicOram, 2025-07-01:
Specifically VFM x is the motor that we're seeing heat build up on

PR #1367 (2025-08-01): Only move the DCM perp if it's outside tolerance
Fixes #1316

See also https://github.com/DiamondLightSource/mx-bluesky/pull/1166

### Instructions to reviewer on how to test:
1. Confirm new test passes

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [  [... cut]
closed 2025-08-01 by DominicOram