DiamondLightSource/dodal #1530: Standard way to define device once that is shared between multiple end statations

oliwenmandiamond, 2025-09-15:
It is common for branched beamlines to have shared components, e.g `PGM` is used by I06 and I06-1.

In dodal, we don't seem to have a standard way of doing this as of yet. At the moment, we have `dodal/beamlines/i05.py, dodal/beamlines/i05_1.py` and `dodal/beamlines/beamline_specific_util/i05_shared.py`. However, `i05.py` and `i05_1.py` both redefine the pgm which defeats the purpose were trying to achieve.

```Python
from dodal.beamline_specific_utils.i05_shared import pgm as i05_pgm

@device_factory()
def pgm() -> PGM:
    return i05_pgm()
```

In other situations, we have I09 defined DCM in two different places when it's the same device `i09.py` and `i09_1.py`. 

This should be defined once in it's own module. This seems to be what I19 have done with `i19_1.py`, `i19_2.py` and `i19_optics.py`, though I think this is slightly different use case as this is a way for access control and has its own BlueAPI service. However, I believe all beamlines, if has shared endstations, should have devices defined once in a shared module. This way, we get BlueAPI to pull devices from beamline `iXX` and `iXX_shared` to be into a single environment e.g

```
worker:
    env:
      sources:
        - kind: planFunctions
          module: sm_bluesky.beamlines.i05.plans
        - kind: planFunctions
          module: sm_bluesky.common.plan_stubs
        - kind: planFunctions
          module: sm_bluesky.common.plans
        - kind: planFunctions
          module: dodal.plans
        - kind: dodal
          module: "dodal.beamlines.i05"
        - kind: dodal
          module: "dodal.beamlines.i05_shared"
```

I think we can also go further and maybe define a `dodal/beamlines/core.py` as well because almost every beamline defines Synchrotron device, we can remove this duplication. 

Finally, as this makes beamlines more modular and we want to test using `dodal connect iXX` for a beamline, I suggest the following change be made to dodal connect:

We define some config in dodal e.g 

```Python
_BEAMLINE_SHARED = {
    "i09": ["i09", "i09_1_shared", "i09_2_shared"]
    ...
}
```
This config defines the whole beamline so when running dodal connect i09, we get:

```
(venv) root ➜ /workspaces/dodal (fix_test_motor_limits_with_new_ophyd_async ✗) 
$ dodal connect i09 -s
====================================================================================================
Attempting connection to i09 (using dodal.beamlines.i09)
2 devices connected (sim mode):
        analyser_driver
        synchrotron
====================================================================================================
Attempting connection to i09_1_shared (using dodal.beamlines.i09_1_shared)
1 devices connected (sim mode):
        dcm
====================================================================================================
Attempting connection to i09_2_shared (using dodal.beamlines.i09_2_shared)
1 devices connected (sim mode):
        pgm
```
We can also add a flag to test the individual beamline only 

```
(venv) root ➜ /workspaces/dodal (fix_test_motor_limits_with_new_ophyd_async ✗) 
$ dodal connect i09 -s -i
====================================================================================================
Attempting connection to i09 (using dodal.beamlines.i09)
2 devices connected (sim mode):
        analyser_driver
        synchrotron
```

## Acceptance Criteria
* Documentation added on standard for shared modules 
* Beamlines conform to standard.
* Dodal connect command adjusted

DominicOram, 2025-09-15:
The i19 solution, documented at https://diamondlightsource.github.io/dodal/main/explanations/decisions/0006-devices-shared-between-endstations.html is partly over-engineered because there is a consistent issue where `19-1` will be running a test and accidentally change the optics device that is currently in use on `i19-2`. In GDA there was no real way to reliably stop this but we have an opportunity to do so with Bluesky. I would suggest before doing this we confirm that the scientists of other branch beamlines see no value in this for their branches. It also appears from https://github.com/DiamondLightSource/dodal/pull/1490 that @DiamondJoseph has some opinions on how this is done so it would be good to get his take.

callumforrester, 2025-09-15:
WRT a core device module, here's some 2 cents from someone who's leaving and has no skin in the game:

I don't like code-as-configuration, I like nicely constrained configuration that isn't turing-complete because you're far less likely to encounter nasty surprises that are hard to debug. I would eventually want all device instantiation to be controlled by YAML/TOML files (or a database) and no user-written Python code at all. It may be useful to think of a solution that doesn't design this out. I don't think this does but I'd appreciate someone else thinking about it.

I have previously suggested a tagging system, so we can tag devices as "facility-wide", "i05-optics-hutch" etc. and then get blueapi to scan and find devices that match a list of tags. Might also be relevant here.

olliesilvester, 2025-09-19:
Just so we have it written down: In yesterday's athena drop-in session it was generally agreed that it's better to have duplication for, eg `i05.py` and `i05_1.py` than it would be to create something like `i05_shared.py`. 

We should make sure everyone is on the same page with that (rather than just those who attended yesterdays session) and then also document that decision

fajinyuan, 2025-09-19:
Thanks @olliesilvester for adding this comment. I think we still need to discuss this matter in next DAQ team meeting. As one of the support team for MMG and SSG, I am strongly disagree with the recommendation you recorded in your comment. Duplication of instances for shared devices for beamlines having multiple end stations and branches will be a nightmere for long term supports as changes to one of end station/branch will not appear in others. We had these issues in current systems, we learnt our lessons, and we would like to not repeat these problems in the new system. 
By default new system should always be better that old system - that is the critical point to carry beamline scientists with us on this journey of migration data acquisition software. Otherwise we will face an unwinable battle to fight.

fajinyuan, 2025-09-22:
Thank you @DiamondJoseph for your update. I think we do need to discuss in the big teams on this issue. I need to understand the evidence and reasons for the suggestion of duplcation of shared devices among end-stations or branches of a single beamline. I have given use cases from our set of beamlines above to give everyone some idea on the different spread of beamline's requirements. I must emphasis these only concerning shared device hardwares within a single beamline, not across beamlines. There might be or will be duplications across beamlines as different beamlines might or will require different specialisation. 

We also need to discuss if there is a better way to specify or deliver beamline customisation or specialisation without duplication in codes.

DiamondJoseph, 2025-09-24:
I've managed to get #1490, where the `set` method of a Movable can be locked behind an unlock predicate, to the point where it is ready for review finally. Highlighting it here for input and comment.

callumforrester, 2025-09-26:
@fajinyuan sorry, when you say inconsistency between branches, do you mean version control branches or physical beamline branch lines?

If the former: Stop running the beamline on a branch for 6+ months, that's the actual bad practice going on in GDA-land. If the latter: Are you sure there's no legitimate reason for two branch lines to have differing configuration for the same underlying optics?

DominicOram, 2025-10-03:
For movable devices. We're currently at the start of a project in MX where we have a device that will be moving between DLS and XFEL. For this we're thinking of having a server sat next to the device that runs the ophyd-devices and some plans. I haven't got strong opinions on how we do this in general except that it's worth remembering we're moving away from a monolith and towards services. We should make sure we think about if we can leverage that